MicrochipTech/cryptoauthtools #6: info.py AssertionError

matriedel, 2018-11-26:
running the info.py example 

sudo ppython info.py -i i2c

I get an AssertionError see the following info output

--------------------------start----------------------------------------

Device Part:
    ATECC508A

Serial number:
    01 23 8B E4 37 8C F6 D2 EE

Configuration Zone:
    01 23 8B E4 00 00 50 00 37 8C F6 D2 EE C0 55 00
    C0 00 AA 00 81 A0 81 A0 81 A0 81 A0 81 A0 81 A0
    81 A0 81 A0 00 00 81 A0 81 A0 81 A0 81 A0 81 A0
    81 A0 81 A0 FF FF FF FF 00 00 00 00 FF FF FF FF
    00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF
    FF FF FF FF 00 00 00 00 FF FF 00 00 00 00 00 00
    33 00 33 00 33 00 33 00 33 00 33 00 33 00 3C 00
    33 00 33 00 33 00 33 00 33 00 33 00 33 00 33 00

Check Device Locks
    Config Zone is locked
    Data Zone is locked

Loading Public key

Traceback (most recent call last):
  File "info.py", line 101, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "info.py", line 85, in info
    assert atcab_get_pubkey(0, public_key) == ATCA_SUCCESS
AssertionError

---------------------stop---------------------------------------------

Do you have any recommendations to solve this issue?
Linux NanoPi-NEO-Air 4.14.52

bryan-hunt, 2018-11-26:
Before a the public key can be obtained from a slot that slot must contain a valid private key. The best way of getting a private key in a slot is to perform a genkey. Looking at the configuration this looks like a custom configuration so I suspect that is the issue (of course you can check that by printing out the return code from atcab_get_pubkey).

We recommend performing the genkey operation on private key slots (before or after "locking" the data zone) when you are programming the devices so as to not have a failure later.

You can see the genkey examples in [config.py](https://github.com/MicrochipTech/cryptoauthtools/blob/a6f03a62fb47fa595b468b3a0ff4bb27c812d8b4/python/examples/config.py#L177)

Also slot 8 is significantly larger than slot 7 if you're looking for a slot to store meta data. From section 2.1:
| Slot | Blocks(1) | Bytes | Bits | Typical Use Notes
| --- | --- | --- | --- | --- |
| 0-7 | 2 | 36 | 288 | Private or Secret Key. Can also be used for data.
| 8 | 13 | 416 | 3328 | Data Reads and Writes can be configured to be restricted in the same manner as all other slots. If this slot is used as a key, then the remaining bytes not required for the secret or private key storage will be ignored. |
| 9-15 | 3 | 72 | 576 | Public Key, Signature or Certificate. For the curve supported by this device, these slots are large enough to contain both the X and Y components of an ECDSA public key or the R and S components of an ECDSA signature. |

matriedel, 2018-11-27:
Thanks for the clarification. So I tried to generate keys by running config.py.
I found two issues:

1) to generate keys only for slot number 7 failed
2) The I2C Address result "Current Address: C0" is not what you get from i2cdetect "60" (see log at the end)

Do you have any hint what causes this behaviour?

------------------------ Log config.py----------------------

sudo python config.py -i i2c

Configuring the device with an example configuration

Serial number:
    01 23 8B E4 37 8C F6 D2 EE

Reading the Lock Status
    Config Zone: Locked
    Data Zone: Locked

Getting the I2C Address
    Current Address: C0

Program Configuration
    Locked, skipping

Activating Configuration
    Already Active

Generating New Keys
    Key 0 Success:
    C4 AD D6 16 97 82 0B 57 9F 36 7B EA B7 8D 15 21
    1F 8E 6E 58 0E 3E E2 47 47 5F D5 A3 5B 07 17 2B
    9B BC 47 A9 BF 78 04 B6 E3 67 8C 0A 71 9B DB FC
    1B C5 55 38 17 76 AD 39 FB 42 C8 0A 4A 03 DC 65

    Key 2 Success:
    F0 C3 13 7C D0 76 4E BC 64 AC B1 75 F2 40 A9 83
    14 01 95 C8 1D E9 E5 CA 06 95 D2 2C 84 56 CD 35
    C8 C1 17 85 45 CE CF 5C DC 59 AE 4D 66 FF BB B2
    90 46 F7 0D D0 BE 34 B0 CA 33 FF 30 E7 20 DF 24

    Key 3 Success:
    D5 39 1B DD 4E 52 7D 52 85 67 58 15 7B 23 98 6A
    75 3E 0C 86 06 44 C5 EC 45 62 10 58 74 79 C2 DD
    B1 26 AD 7B 30 E8 9F 79 8C 64 BC 53 73 D9 2C C0
    CD F4 56 9E 96 8E DF 48 4D 10 95 C6 06 66 58 F6

Traceback (most recent call last):
  File "config.py", line 214, in <module>
    configure_device(args.iface, args.device, args.i2c, args.gen, **parse_interface_params(args.params))
  File "config.py", line 198, in configure_device
    assert ATCA_SUCCESS == atcab_genkey(7, pubkey)

------------------------ End Log config.py----------------------

------------------------  Log i2cdetect----------------------
sudo i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

bryan-hunt, 2018-11-27:
1) As per the original comment you have a highly custom configuration. You've set up slot 7 to be completely inaccessible. Have you worked with an FAE to determine use cases and configuration?
2) The cryptoauthentication devices use the upper 7 bits to identify the address while the convention is to use the lower 7 bits. When you left shift the i2cdetect value 0x60 is 0xC0 (device convention).

matriedel, 2018-11-27:
1) Our scenario is realizing a secure AWS IoT cloud communication based on a Linux system running Python and an ATECC508A connected via the I2C bus. In the past talking to an FAE, there was no direct SW support for I2C communication directly from a Linux system.  With your code, it seems this has changed now. Now we have to learn how to realize AWS JITR using the ATECC508A or ATECC509A chipset and the library.
2) Got it - thanks

bryan-hunt, 2018-11-27:
This is a problematic use case since python core does not support anything but filesystem private keys and the ssl module is not very pythonic so it does not offer any extensibility in credentials (it is just a thin wrapper over OpenSSL). See the [python bug tracker](https://www.bugs.python.org/issue28695) for more details.

Your options will be:

a) Use a custom authorizer [see AWS docs](https://docs.aws.amazon.com/iot/latest/developerguide/iot-custom-authentication.html)
b) A third party python module that replaces the core ssl module. In order to use cryptoauthlib as a smartcard interface you'll have to download and build its as a PKCS11 library. I have not done any evaluation of the alternative TLS implementations available for python. See the [pkcs11 readme](https://github.com/MicrochipTech/cryptoauthlib/blob/pkcs11/app/pkcs11/README.md)

matriedel, 2018-11-27:
Hi Bryan

 

What we achieved so far is to implement the AWS JITR to register an IoT node and using mainly authorized MQTT operations.

 

But the key pairs are currently stored in the file system of our Linux based IoT node.

What we wanted to achieve is to move the keys in a secure way to the encryption chipset

 

So far we tried to use eclet to communicate with the encryption chipset. 

 <https://github.com/cryptotronix/EClet> https://github.com/cryptotronix/EClet

The only usable operation we could achieve was reading the serial number.

 

It seems when testing eclet we also locked accidentally the config and data zone.

So I need to organize and solder a new chip on our prototype.

 

It is already a long journey to achieve a reasonable implementation and some developers already gave up.

 

Many thanks for your comments

 

Best

 

Matthias

closed 2019-01-02 by bryan-hunt