MicrochipTech/cryptoauthtools #9: After device is provisioned communication gets unreliable

matriedel, 2019-03-18:
For product support you must talk to your assigned FAE or submit a request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). Product support questions can't be answered here.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case)

bryan-hunt, 2019-03-18:
This would be a question for the vendor of your host board.

matriedel, 2019-03-18:
Well, as long ATECC608A is not provisioned everything works as expected.
Once ATECC608A is provisioned the issues are noticeable. In both cases, the HW keeps the same.
As a result, my conclusion would be I have chosen a specific ATECC608A wrong parameter setting.

bryan-hunt, 2019-03-18:
Hardware/product support questions must go through your assigned FAE or submitted as a request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). 

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case).

closed 2019-03-21 by matriedel