MicrochipTech/cryptoauthtools #10: How to do a CSR / Signing / apply Certificate Flow?

pfried, 2019-06-12:
First of all thank you for the library, it has been a great help getting started with the Crypto provisioning.

I can see that there is a CSR example which is alright, but after signing the Certificate one would need to reupload that signed certificate back to the device. It would be great if a fulll flow of steps could be shown in an example (config.py already does a lot, why not extend that a bit)

Also I find it hard to find good documentation of the slot system, Using the library would be easier if there was a brief introduction into the slots and an example how to use them in a  typical IoT (AWS, Google, etc.) scenario

pfried, 2019-06-12:
Also I want to note that converting *.pem files (pem -> compressed ec pub) to the internal format is necessary when receiving the signed certificates from external sources. Is there an existing helper function? Otherwise this would be helpful

bryan-hunt, 2019-06-12:
Using CSRs is not process which is secured cryptographically so generally we consider it for demonstration purposes only - once you have the certificate you would use directly. The cryptoauth devices are not smart cards and fit a very wide variety of use cases so how you configure them is directly related to your needs. 

There are a wealth of information about these devices and reference implementations available on the [Microchip website](https://www.microchip.com/design-centers/security-ics/cryptoauthentication)

Besides the written content there is also a youtube series:
https://www.youtube.com/watch?v=TdexOLD33bs&list=PL9B4edd-p2aio5a0N8mVYM2cYqQW0OUgp

Example Cloud IoT Uses:

Amazon: https://www.microchip.com/design-centers/security-ics/cryptoauthentication/cloud-authentication/aws-iot-atecc608a

Google: https://www.microchip.com/design-centers/security-ics/cryptoauthentication/cloud-authentication/google-iot-core-atecc608a

Greengrass HSI: https://www.microchip.com/design-centers/security-ics/cryptoauthentication/cloud-authentication/aws-greengrass-atecc608a

pfried, 2019-06-12:
@bryan-hunt Thank you for responding and for the provided links.

I do not understand the follwing sentence:

> Using CSRs is not process which is secured cryptographically so generally we consider it for demonstration

I want to lay out my thoughts and the current process I did:

1. Create a configuration and write it to the device
2. Lock the config
3. Generate ecc keypairs on the device
4. Generate a signed CSR from the device
5. Create a certificate in the AWS console with the CSR in a PEM format (in other examples this is done directly with a signer, but for a first flow I would like to avoid this step)
6. Write back the certificate to the device (this step is currently missing)
7. Verify and lock

I could program the certificate to the device, but as I want to have one firmware which is able to pull its credentials from the crypto chip I thought it is necessary to write that back to the device.

Maybe I am wrong but to me it feels this is a tooling issue since I need to write back the certificate to the device from a PEM file and that is what I asked about.

About the AWS use case: My issue is that it requires a certain hardware and uses some serial protocol from the python side to the programmer, so I have difficulties following it along because the code is basically split into multiple components

Best, Friedrich

bryan-hunt, 2019-06-12:
I suggest following the Amazon IoT example:  https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit

The only cryptographic operation done to validate a CSR is that the signature matches the public key of the CSR. That does not give you an ability to verify a chain of trust. That is why it is only safe for demonstrations and not for production. A production system needs an established chain of trust with proper key ceremonies done for the root. This is a normal PKI process.

The ATECC608A has limited storage space and will not accept a full certificate. In order to save the important data of the certificate a mapping needs to be created (the certificate defintion - atcacert_def) which is then used by the library's certificate API to read and write certificates.

pfried, 2019-06-12:
> The only cryptographic operation done to validate a CSR is that the signature matches the public key of the CSR. That does not give you an ability to verify a chain of trust. That is why it is only [...]

Okay, I think I understood your point of missing trust where a CSR is blindly signed and is therefore not suitable for production.

We are just evaluating the ATECC608A device and got the dev kit and we want to see how it works with our (custom board) host MCU and mbedtls tooling this is why I just wanted to get things going by writing some custom python app to provision the dev kit.

> The ATECC608A has limited storage space and will not accept a full certificate. In order to save the important data of the certificate a mapping needs to be created (the certificate defintion - atcacert_def) which is then used by the library's certificate API to read and write certificates.

Yes I came along the `atcacert_def` already for the csr and I understand its a template combined with the  compressed data in the chip makes up the certificate. So it comes down to that I still need to extract the compressed data part from the PEM format.

C-Code from the firmware:
https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit/blob/master/firmware/SAMG55/AWS_IoT_Zero_Touch_SAMG55/src/provisioning_task.c#L830

Python part:
https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit/blob/master/kit_provision.py#L126

I will see what I can extract from it, I hoped there was some code available, but  I will have to dig in a bit deeper.

bryan-hunt, 2019-06-12:
This gist might help you out. The definitions used are provided and can be included in an of the C applications using cryptoauthlib.

https://gist.github.com/bryan-hunt/a6c27e99391213e462ac6fc58ed84929

pfried, 2019-06-13:
Great resource, thanks a lot for helping! I will close the issue as I think I can follow the steps

closed 2019-06-13 by pfried