MicrochipTech/cryptoauthtools #11: Soft reset to make I2C address effective

matriedel, 2019-06-21:
For product support you must talk to your assigned FAE or submit a request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). Product support questions can't be answered here.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case)

matriedel, 2019-06-21:
During device configuration, I am changing the I2C address of the chipset.  I currently have to switch off and on the device to take effect of the new I2C address. I would like to avoid this system reboot cycle.

Is there a way in python to soft reset the chipset e.g. via the Sleep Flag that transitions the ATECC508A to the low power sleep mode, which causes a complete reset of the device?

To put the device in a low power sleep mode via a python call would be appreciated in any case.

Thank you for your advice.

bryan-hunt, 2019-06-25:
atcab_release will do what you are asking. The device does not immediately go to sleep when issued the sleep command however so you should wait before attempting to reconnect on the new i2c address. In python time.sleep(1) is more than sufficient even though the specification is less than 20ms.

matriedel, 2019-06-25:
O.K thank you for your advice

closed 2019-06-25 by matriedel