MicrochipTech/cryptoauthtools #12: Make config in the examples more readable

pfried, 2019-06-25:
The configuration throughout the examples are bytearrays hard coded. As I think this repository shows examples on how to use the library at least the `config.py` example should use a better format or comment.

I would love to see a more adaptable (or say user changable) code like:

```
config = Atecc608aConfig()
config.I2C_Address = 0xC0

config.CountMatch.UserExtraAdd = 1

# Make these bits unlocked to be able to validate when writing the config
config.LockValue = 0x55
config.LockConfig = 0x55

keySlots = [0, 1]

for index, slot_config in enumerate(config.SlotConfig):
	# Private key in slot 0
	if index in keySlots:
		slot_config.IsSecret = 1
		slot_config.ReadKey = 5

for index, key_config in enumerate(config.KeyConfig):
	if index in keySlots:
		key_config.Private = 1
		key_config.PubInfo = 1
		# P256 NIST ECC key
		key_config.KeyType = 4
		key_config.ReqRandom = 1
````

bryan-hunt, 2019-06-25:
These examples are to illustrate how to utilize the library. Several hundred lines of configuration is not helpful to that end. Config.py sets up devices with a tested and vetted configuration that is suitable for the most common IOT use cases - it is not a general purpose configuration application.

It is easy to jump back and forth between the bytearray (which is what the API expects) and the python representation of it for those who would like to explore it.

pfried, 2019-06-25:
I do not agree, for me getting started with the library was not an ideal experience since one has to do quite some things to use the chip to connect to AWS (or any other cloud service) where you will need a configuration which matches your specific use case (Your code helped a lot with the signing part). Also in a typical development you would try features step by step (e.g. using encrypted read/write operations are needed for production, but for development it might be an obstacle)

> Config.py sets up devices with a tested and vetted configuration

This is the config I used, and the first thing you will have to do is to decrypt the config to see if it works for your own project. This is where the snippet above could have helped. Of course you can expect people to read the datasheet but the bytearrays are practically unreadable (bitfields etc., the information is distributed over several datasheets (508A-608A).

I found no guide on how the slots are configured and how they are intended to be used in a specific scenario (something like Slot 0 is for the private key, Slot 10 for the device certificate, Slot 10 for the signer public key, slot 11 for the CA certificate, ...)

As this repository shows examples I see no problem in showing twenty lines of explicit written configuration.

I ended up buying some more chips since the config from the example does not work with our setup (it almost did).

closed 2022-02-11 by github-actions[bot]