MicrochipTech/cryptoauthtools #13: Can't run info.py

seungwhanrhee, 2019-09-03:
Hi,
Figured this is the best location to ask this question but please let me know if not.

Tried both on macOS 10.14.4 and Windows 10 64bit.
Python : 2.17.15 on both OS.
AT88CK590 plugged into the USB port of PC.

I ran
pip install cryptoauthlib
pip install -r requirements.txt

And I get this.
$ python info.py
Traceback (most recent call last):
  File "info.py", line 96, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "info.py", line 48, in info
    assert atcab_init(cfg_ateccx08a_kithid_default()) == ATCA_SUCCESS
AssertionError

Thanks advance for your help.

bryan-hunt, 2019-09-03:
This seems like it might be a USB problem. If you are using a hub or docking station can you try it ina  USB port that is directly on the computer?

seungwhanrhee, 2019-09-03:
Thanks for the quick reply Bryan,
But unfortunately your suggestion didn't help.
Tried directly to the computer, different hub, and different PC.
All same.

bryan-hunt, 2019-09-03:
Okay I've investigated and this is a bug in the updated version of cryptoauthlib. If you have pip install version 20190517 of cryptoauthlib it'll work with the CK590. 

I'm working on a fix but I don't know when it will be published yet.

seungwhanrhee, 2019-09-03:
That worked, thanks!

By the way, I don't have DM320109, but I have AT88CKECCSIGNER kit(3 dongles for $100) which turns out to be same as AT88CK590($20 ea)!

kpm-at-hfi, 2020-06-29:
@bryan-hunt can you provide any update concerning the fix you mentioned above?

dexterac, 2022-02-07:
I get this error while working with i2c connected device
`cryptoauthlib.exceptions.UnsupportedInterface: The selected interface specified by cfg_ateccx08a_i2c_default is not supported by this version of the library
`

bryan-hunt, 2022-02-10:
This has long since been corrected.

closed 2022-02-10 by bryan-hunt

maxall41, 2022-09-26:
Actually, this still seems to be an issue I was getting the same error as @dexterac until I switched to 20190517 which fixed it.

op, 2022-10-13:
I silenced this error in main by adding -DATCA_HAL_I2C=1 to cmake. (If you add -DATCA_PRINTF=1, you can quite easily trace your problem down.)