MicrochipTech/cryptoauthtools #19: ATSHA204A on Raspberry Pi I2C walkthrough.

krish2487, 2020-03-10:
**Issue**
Hello all, I undertand that the cryptotools repo has some python examples for configuring / using the crypto devices. We are trying to use / learning to use the ATSHA204A chip for a public / private key pair generation. The ATSHA chip is connected directly to the I2C 3V3, GND, SDA, SCL of the raspberry pi. We are using the latest release of the repo for testing. I see from one of the issues [here](https://github.com/MicrochipTech/cryptoauthtools/issues/13) that we will need to roll back to an older version of the cryptoauthlib for the examples to work, which we have done as well.

**To Reproduce**
1. Connect the ATSHA as shown.
2. Run 'sudo i2cdetect -y 1' on the commandline of the Pi.
3. Check the reported device address and shift the address by 1 position.

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

```
64 is the actual device here. C8 is the shifted address.
2. Run the command ' python config.py -i i2c -d sha -p slave_address=0xC8' to configure the device.
```
Configuring the device with an example configuration

ATSHA204A
0

Serial number:
    01 23 FE E3 C9 15 FF D5 EE

Reading the Lock Status
    Config Zone: Locked
    Data Zone: Locked

Getting the I2C Address
    Current Address: C8

Program Configuration
    Locked, skipping

Activating Configuration
    Already Active

Generating New Keys

Device Successfully Configured

```

This is what I see above. I already ran the same step and configured the ATSHA, hence I am seeing the "locked" message.
However, I understand that when the device geenrates and writes new keys, it must print them to the screen as Key 0 success ,1,2 and so on.. But I dont see any of that output here.

3. Running 'python info.py -i i2c -d sha -p slave_address=0xC8' gives this error.

```
Device Part:
    ATSHA204A

Serial number:
    01 23 FE E3 C9 15 FF D5 EE

Configuration Zone:
    01 23 FE E3 00 09 04 00 C9 15 FF D5 EE 10 01 00
    C8 00 55 00 8F 80 80 A1 82 E0 C4 F4 84 00 A0 85
    86 40 87 07 0F 00 C4 64 8A 7A 0B 8B 0C 4C DD 4D
    C2 42 AF 8F FF 00 FF 00 FF 00 FF 00 FF 00 FF 00
    FF 00 FF 00 FF FF FF FF FF FF FF FF FF FF FF FF
    FF FF FF FF 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Check Device Locks
    Config Zone is locked
    Data Zone is locked

Loading Public key

Traceback (most recent call last):
  File "info.py", line 101, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "info.py", line 86, in info
    assert atcab_get_pubkey(0, public_key) == ATCA_SUCCESS
AssertionError

```

I expect to see the public keys being printed to the screen. However it fails in the specific assertion.

**Expected behavior**
 I expect that the script 
a) running config.py Prints the messages "Key 0 Success" and so on during the config stage and prints the keys to the screen.
b) running info.py reads the device and extracts / generates the public keys and prints them on the screen.

**Additional context**
I suspect that the script is not really doing anything for the ATSHA204 chip and is geared towards the remaining family members. 

I would like to know if there are any gaps in my understanding of how to use the scripts.. and if I am doing a step wrong or incomplete.. 
If not, then is there anything I can do to help or request help to get a working example for the ATSHA204 on the Pi?

Thanks in advance!!

bryan-hunt, 2020-03-10:
The ATSHA204A does not contain any public/private cryptography. If you need asymmetric cryptography the correct part to be using is the ATECC608A which uses Elliptic Curve Cryptography (SECP256r1)

krish2487, 2020-03-10:
@bryan-hunt 
Thank you for the prompt response bryan. 
If it is not too much trouble. Can you briefly give a walkthrough of how one would go about using the ATSHA204A?

How can I create a digital signature using the ATSHA204A? 
How can I verify it on a host??

I apologize if it seems like asking too much.. I am still not clear on how and where the ATSHA204A can or should be used??

Thank you and have a nice day!!

bryan-hunt, 2020-03-10:
For product support you must talk to your assigned FAE or submit a request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). Product support questions can't be answered here.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case)

krish2487, 2020-03-10:
@bryan-hunt 
Thank you anyway! :-)
It has been helpful and informative.

Thank you and have a nice day!!

closed 2020-03-10 by krish2487