MicrochipTech/cryptoauthtools #31: config.py throwing exception when using an ATECC680A

dexterac, 2022-02-10:
**Describe the bug**
A clear and concise description of what the bug is.

config.py with atecc608a connected to raspberrypi with I2C connector fails with error message : `NameError: name 'Atecc608Config' is not defined`

Details of the error:

`
$ python info.py -i i2c

Device Part:
    ATECC608A

Serial number:
    01 23 B2 01 E3 68 71 FF EE

Configuration Zone:
    01 23 B2 01 00 00 60 02 E3 68 71 FF EE C1 71 00
    C0 00 00 00 83 20 87 20 8F 20 C4 8F 8F 8F 8F 8F
    9F 8F AF 8F 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 AF 8F 00 00 FF FF 00 00 00 00 FF FF FF FF
    00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 00 00 00 00 55 55 FF FF 00 00 00 00 00 00
    33 00 33 00 33 00 1C 00 1C 00 1C 00 1C 00 1C 00
    3C 00 3C 00 3C 00 3C 00 3C 00 3C 00 3C 00 1C 00

Check Device Locks
    Config Zone is unlocked
    Data Zone is unlocked

Done

$ python config.py -i i2c

Configuring the device with an example configuration

Serial number:
    01 23 B2 01 E3 68 71 FF EE

Reading the Lock Status
    Config Zone: Unlocked
    Data Zone: Unlocked

Getting the I2C Address
    Current Address: C0

Program Configuration

    New Address: 6A
    The AT88CK590 Kit does not support changing the I2C addresses of devices.
    If you are not using an AT88CK590 kit you may continue without errors
    otherwise exit and specify a compatible (0xC0) address.
    Continue (Y/n): Y
    Programming ATECC608A Configuration
        Success
    Verifying Configuration
        Success
    Locking Configuration
        Locked

Activating Configuration
Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/config.py", line 241, in <module>
    configure_device(args.iface, args.device, args.i2c, args.gen, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/config.py", line 172, in configure_device
    key_gen(dev_name)
  File "/home/pi/cryptoauthtools/python/examples/config.py", line 201, in key_gen
    config = Atecc608Config.from_buffer(config_data)
NameError: name 'Atecc608Config' is not defined
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --                        
$ python info.py -i i2c

Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 93, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 50, in info
    assert atcab_info(info) == ATCA_SUCCESS
AssertionError

`

**To Reproduce**
Steps to reproduce the behavior:
1. Install Raspios Bullseye 32bit for Raspberry Pi 4 : https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/2022-01-28-raspios-bullseye-armhf-lite.zip
2. Install Cryptoauthlib and cryptoauthtools python packages using pip.
3. Install i2ctools using apt
4. Connect atecc608a board and verify with i2cdetect command and info.py script (as shown in output above)
5. run python config.py -i i2c to configure 

This is my 4th attempt to configure and same error while trying with different versions of OS.

bryan-hunt, 2022-02-10:
I am not able to reproduce this error.

1) cryptoauthtools can't be installed via pip from an official repo - if you are installing it then you're getting it from a bad index and you should remove it from your configuration.
2) cryptoauthlib.device has exported this object for a very long time.

I have cryptoauthlib version 20211006 installed (python3 -m pip list) and have cloned the latest version of cryptoauthtools

dexterac, 2022-02-10:
Apologies. I had cloned cryptoauthtools repo.

To Reproduce
Steps to reproduce the behavior:

1. Install Raspios Bullseye 32bit for Raspberry Pi 4 : https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/2022-01-28-raspios-bullseye-armhf-lite.zip
2. Install cryptoauthlib python package using pip
3. git clone https://github.com/MicrochipTech/cryptoauthtools.git
3. Install i2ctools using apt
4. Connect atecc608a board and verify with i2cdetect command and info.py script (as shown in output above)
5. run python config.py -i i2c to configure

dexterac, 2022-02-10:
When I upgrade cryptoauthlib to 20211006 info.py fails with this error message : 

```
python info.py -i i2c
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/cryptoauthlib/iface.py", line 160, in _iface_load_default_config
    return ATCAIfaceCfg.in_dll(get_cryptoauthlib(), name)
ValueError: /lib/arm-linux-gnueabihf/libcryptoauth.so: undefined symbol: cfg_ateccx08a_i2c_default

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 93, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 32, in info
    cfg = eval('cfg_at{}a_{}_default()'.format(atca_names_map.get(device), atca_names_map.get(iface)))
  File "<string>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/cryptoauthlib/iface.py", line 167, in cfg_ateccx08a_i2c_default
    return _iface_load_default_config('cfg_ateccx08a_i2c_default')
  File "/home/pi/.local/lib/python3.9/site-packages/cryptoauthlib/iface.py", line 162, in _iface_load_default_config
    raise UnsupportedInterface('The selected interface specified by {} is not supported by this version of the library'.format(name))
cryptoauthlib.exceptions.UnsupportedInterface: The selected interface specified by cfg_ateccx08a_i2c_default is not supported by this version of the library
```

bryan-hunt, 2022-02-10:
You didn't build the library with I2C enabled.

dexterac, 2022-02-10:
> You didn't build the library with I2C enabled.

I followed these steps: https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup

bryan-hunt, 2022-02-10:
If you build and install a library you have to configure it properly. It looks like you have a system copy as well as the one installed with the python wheel. The python module will default to the system copy if it finds one - thus your system copy has to be configured properly. Configuration options are here:

https://github.com/MicrochipTech/cryptoauthlib/wiki/Configuration

dexterac, 2022-02-10:
That helped to get the device detected with p11tools!

```
$ p11tool --list-all 'pkcs11:model=ATECC608A'
Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=23B201E36871FFEE;token=00ABC;object=device;type=private
	Type: Private key (EC/ECDSA-SECP256R1)
	Label: device
	Flags: CKA_PRIVATE; CKA_SENSITIVE;
	ID:

Object 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=23B201E36871FFEE;token=00ABC;object=device;type=public
	Type: Public key (EC/ECDSA-SECP256R1)
	Label: device
	ID:

i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- 35 -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

python info.py -i i2c

Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 93, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 50, in info
    assert atcab_info(info) == ATCA_SUCCESS
AssertionError
```

bryan-hunt, 2022-02-10:
You have to provide the i2c address to the python scripts. In other words:

```
python info.py -i i2c -p address=6a
```

dexterac, 2022-02-10:
> You have to provide the i2c address to the python scripts. In other words:
> 
> ```
> python info.py -i i2c -p address=6a
> ```

```
python info.py -i i2c -p address=6a

Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 93, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 50, in info
    assert atcab_info(info) == ATCA_SUCCESS
AssertionError
```

dexterac, 2022-02-10:
Second attempt it worked: 
```
$ python info.py -i i2c -p address=6a

Device Part:
    ATECC608

Serial number:
    01 23 B2 01 E3 68 71 FF EE

Configuration Zone:
    01 23 B2 01 00 00 60 02 E3 68 71 FF EE C1 71 00
    6A 00 00 01 85 00 82 00 85 20 85 20 85 20 C6 46
    8F 0F 9F 8F 0F 0F 8F 0F 0F 0F 0F 0F 0F 0F 0F 0F
    0D 1F 0F 0F FF FF FF FF 00 00 00 00 FF FF FF FF
    00 00 00 00 00 00 03 F7 00 69 76 00 00 00 00 00
    00 00 00 00 00 00 55 00 FF FF 0E 60 00 00 00 00
    53 00 53 00 73 00 73 00 73 00 38 00 7C 00 1C 00
    3C 00 1A 00 3C 00 30 00 3C 00 30 00 12 00 30 00

Check Device Locks
    Config Zone is locked
    Data Zone is unlocked

Done
```

bryan-hunt, 2022-02-10:
The device is still "unlocked". Only the configuration has been programmed and locked. The locks are described in the datasheet. The config zone lock makes the config zone immutable. The data zone lock activates the configuration so whatever configuration is specified is enforced. The data zone lock does not make the data zone immutable - that is controlled by the configuration and the slot locks.

dexterac, 2022-02-10:
I'm unable to create new keys, kindly suggest alternate methods to generate keys

```
$ python config.py -i i2c -p address=6a

Configuring the device with an example configuration

Serial number:
    01 23 B2 01 E3 68 71 FF EE

Reading the Lock Status
    Config Zone: Locked
    Data Zone: Unlocked

Getting the I2C Address
    Current Address: 6A

Program Configuration
    Locked, skipping

Activating Configuration
Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/config.py", line 241, in <module>
    configure_device(args.iface, args.device, args.i2c, args.gen, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/config.py", line 172, in configure_device
    key_gen(dev_name)
  File "/home/pi/cryptoauthtools/python/examples/config.py", line 203, in key_gen
    raise ValueError('Unsupported device {}'.format(dev_name))
ValueError: Unsupported device ATECC608
```

bryan-hunt, 2022-02-10:
Reclone the cryptoauthtool repo

dexterac, 2022-02-11:
> Reclone/update the cryptoauthtool repo. I pushed the fix for #30 an hour ago.

```
$ python config.py -i i2c -p address=6a

--- WARNING - THIS IS FOR DEMONSTRATION NOT FOR ACTUAL CONFIGURATION ---
The configuration this script will program is for demonstration purposes only

You should use the Trust Platform Development Suite for device configuration

  https://www.microchip.com/en-us/products/security-ics/trust-platform#Getting%20Started

Would you like to continue executing this script?
  Continue (Y/n): Y

Configuring the device with an example configuration

Serial number:
    01 23 B2 01 E3 68 71 FF EE

Reading the Lock Status
    Config Zone: Locked
    Data Zone: Unlocked

Getting the I2C Address
    Current Address: 6A

Program Configuration
    Locked, skipping

Activating Configuration
    Generating key pair in slot 0
-----BEGIN PUBLIC KEY-----
"adadahdlkasjdhalskdjhasldhaldhalkdhalkdhalkdhaldadhaldhalkdha"
-----END PUBLIC KEY-----
    Generating key pair in slot 1
-----BEGIN PUBLIC KEY-----
"adadahdlkasjdhalskdjhasldhaldhalkdhalkdhalkdhaldadhaldhalkdha"
-----END PUBLIC KEY-----
    Generating key pair in slot 2
-----BEGIN PUBLIC KEY-----
"adadahdlkasjdhalskdjhasldhaldhalkdhalkdhalkdhaldadhaldhalkdha"
-----END PUBLIC KEY-----
    Generating key pair in slot 3
-----BEGIN PUBLIC KEY-----
"adadahdlkasjdhalskdjhasldhaldhalkdhalkdhalkdhaldadhaldhalkdha"
-----END PUBLIC KEY-----
    Generating key pair in slot 4
-----BEGIN PUBLIC KEY-----
"adadahdlkasjdhalskdjhasldhaldhalkdhalkdhalkdhaldadhaldhalkdha"
-----END PUBLIC KEY-----
    Activated

Device Successfully Configured
```

dexterac, 2022-02-11:
@bryan-hunt You've been a great help!!!!

I realized the small issue with documentation where the steps to follow while working with i2c devices isn't clearly documented. Another issues which could be common with i2c interface is the address likely to change while configuring the board.

When I started the board reported on address 0x60 and while configuring it changed to 0x35, I missed the step to build with i2c support so that could be root cause of issue, however the board can still be reconfigured based on the steps you helped me with.

Thanks for your help!

bryan-hunt, 2022-02-11:
Marking and resolved and closing.

closed 2022-02-11 by bryan-hunt