MicrochipTech/cryptoauthtools #34: NameError: name 'setup_example_runner' is not defined

thecyberdotninja, 2022-08-07:
Just getting started with the ATECC508A connected to a Raspberry Pi.

I get the following error when trying to run info.py
`
pi@raspberrypi:~ $ python info.py 
Traceback (most recent call last):
  File "/home/pi/info.py", line 90, in <module>
    parser = setup_example_runner(__file__)
NameError: name 'setup_example_runner' is not defined
`

Any help would be greatly appreciated.

Thanks!

bryan-hunt, 2022-08-07:
I would suggest running from a cloned repo rather than as copied files.

thecyberdotninja, 2022-08-07:
Got it, new problem:

pi@raspberrypi:~/cryptoauthtools/python/examples $ python info.py 
Traceback (most recent call last):
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 93, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthtools/python/examples/info.py", line 45, in info
    assert atcab_init(cfg) == ATCA_SUCCESS
AssertionError

bryan-hunt, 2022-08-07:
If you're using i2c then you'll need to execute the script with the proper options. If you're using usb then check out the udev rules: https://github.com/MicrochipTech/cryptoauthlib/wiki/Udev-Rules

thecyberdotninja, 2022-08-07:
Yeah, I just figured it out, thanks.  I am using i2c and had to use

`python info.py -i i2c`

and it worked fine.  

My ultimate goal is to verify signatures from the ATECC608A using gpg2 and the ATECC608A's public key, but I'm struggling with getting the data in the correct format.  Is there a walkthrough somewhere that could help?

Information overload at cryptoauthlib on microchip's website...

Thanks for the support!

closed 2022-08-07 by thecyberdotninja

bryan-hunt, 2022-08-08:
@thecyberdotninja The device emits the following values:

* The public key is returned as the big integers x & y which are the affine coordinates
* The signature is returned as the big integers r & s which are the x coordinate and signature proof respectively.

When used in a crypto system you will likely need to identify the EC curve group or name if you wanted to use them directly otherwise it would require converting to an ASN.1 representation of them. There a number of ways to go about this. If you would prefer to skip to something that "just works" then you'll have to use the pkcs11 api per https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup

With a bit of research however it is generally not too difficult to provide the public key and signature integers to the crypto library directly with the proper structure initialization.